xreef/EByte_LoRa_E22_Series_Library #1: Compile Error for STM32 Core for Arduino

momonator25, 2021-03-22:
Hello I know this libary is not supposed to work with STM32 but it does with a little fix.
Right now there is a conflict in naming:
`statesNaming.h:20:13: error: 'SUCCESS' conflicts with a previous declaration 20 | SUCCESS = 1,
`
with
`stm32f1xx.h:166:3: note: previous declaration 'ErrorStatus SUCCESS' 166 | SUCCESS = 0U,
`

The compile error can be avoided if SUCCESS gets renamed.

xreef, 2021-03-23:
Hi momonator25,
thanks for the alert, I'm going to fix as soon as possible (but I need some time).
Thanks again Renzo

momonator25, 2021-03-23:
> 
> 
> Hi momonator25,
> thanks for the alert, I'm going to fix as soon as possible (but I need some time).
> Thanks again Renzo

No Problem and thanks for your fast reply! Don't hurry! Best regards

xreef, 2021-05-24:
Hi momonator25,
I test e22 with STM32 ad arduino IDE and I haven't that problem, I had another problem with softwareserial and I'm going to fix It.
Witch platform do you use?
Bye Renzo

lejibxl, 2022-02-15:
Dear xreef,
I encountered the same problem with a Nucleo-32 G031K8 card.
I renamed in file "statesNaming.h" "SUCCESS" term  to "SUCCESS_E22" and it works again
Thanx for your excellent  library!

closed 2022-04-11 by xreef