xreef/EByte_LoRa_E22_Series_Library #3: Warning : Save mode returned not recognized!

WillingLearner, 2021-05-31:
Hello, I set up esp32 and e22 400t 20d model  as described and I open setConfiguration example I upload code. I see "save mode returned not recognized" warning on the serial monitor. What can i do to fix it ?

xreef, 2021-05-31:
Hi WillingLearner,
It isn't a string related to my library, I don't know. Try to give more info.
Bye Renzo

WillingLearner, 2021-05-31:
I search a little bit about it, I found that in /include/stateNaming.h in the line of 70 like

	  case ERR_E22_HEAD_NOT_RECOGNIZED:		return F("Save mode returned not recognized!");		break;

xreef, 2021-05-31:
Damn, Yes It's true, give me some information and the correct model, e22 400t 20d not in list at EByte.
Bye Renzo

WillingLearner, 2021-05-31:
I am sorry, E22400T22D model

xreef, 2021-06-01:
Hi,
the returned data was wrong, enable debug and try a get configuration.
Bye Renzo

WillingLearner, 2021-06-02:
How to enable debug ? I tried to get configuration, serial monitor just says:

Success
1
Success
1

Model no:22
Version : 10
Features : 0

xreef, 2021-06-09:
Hi,
Yes the library is fully working with esp32.

The output you posted isn't getConfiguration.

You must uncomment LoRa_E22_DEBUG define.

Bye Renzo

WillingLearner, 2021-06-11:
I uncommented debug line and managed to get configuration as below :

`RX MIC ---> -1
TX MIC ---> -1
AUX ---> 18
M0 ---> 21
M1 ---> 19
Init AUX pin!
Init M0 pin!
Init M1 pin!
Begin ex
Begin Hardware Serial
Begin 
MODE NORMAL!
AUX HIGH!
Complete!
MODE CONFIGURATION!
AUX HIGH!
Complete!
3
Available buffer: 12 structure size: 12
AUX HIGH!
Complete!
----------------------------------------
HEAD : C1 0 9
 
AddH : 0
AddL : 3
NetID : 0
 
Chan : 23 -> 433MHz
 
SpeedParityBit     : 0 -> 8N1 (Default)
SpeedUARTDatte     : 11 -> 9600bps (default)
SpeedAirDataRate   : 10 -> 2.4kbps (default)
 
OptionSubPacketSett: 0 -> 240bytes (default)
OptionTranPower    : 0 -> 22dBm (Default)
OptionRSSIAmbientNo: 1 -> Enabled
 
TransModeWORPeriod : 11 -> 2000ms (default)
TransModeTransContr: 0 -> WOR Receiver (default)
TransModeEnableLBT : 0 -> Disabled (default)
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeEnabRepeat: 0 -> Disabled (default)
TransModeFixedTrans: 1 -> Fixed transmission (first three bytes can be used as high/low address and channel)
----------------------------------------
MODE NORMAL!
AUX HIGH!
Complete!
Success
1
----------------------------------------
HEAD : C1 0 9
 
AddH : 0
AddL : 3
NetID : 0
 
Chan : 23 -> 433MHz
 
SpeedParityBit     : 0 -> 8N1 (Default)
SpeedUARTDatte     : 11 -> 9600bps (default)
SpeedAirDataRate   : 10 -> 2.4kbps (default)
 
OptionSubPacketSett: 0 -> 240bytes (default)
OptionTranPower    : 0 -> 22dBm (Default)
OptionRSSIAmbientNo: 1 -> Enabled
 
TransModeWORPeriod : 11 -> 2000ms (default)
TransModeTransContr: 0 -> WOR Receiver (default)
TransModeEnableLBT : 0 -> Disabled (default)
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeEnabRepeat: 0 -> Disabled (default)
TransModeFixedTrans: 1 -> Fixed transmission (first three bytes can be used as high/low address and channel)
----------------------------------------
MODE CONFIGURATION!
AUX HIGH!
Complete!
AUX HIGH!
Complete!
Clear buffer...ok!
Available buffer: 0 structure size: 12
----------------------------------------
HEAD : C0 0 9
 
AddH : 0
AddL : 3
NetID : 0
 
Chan : 23 -> 433MHz
 
SpeedParityBit     : 0 -> 8N1 (Default)
SpeedUARTDatte     : 11 -> 9600bps (default)
SpeedAirDataRate   : 10 -> 2.4kbps (default)
 
OptionSubPacketSett: 0 -> 240bytes (default)
OptionTranPower    : 0 -> 22dBm (Default)
OptionRSSIAmbientNo: 1 -> Enabled
 
TransModeWORPeriod : 11 -> 2000ms (default)
TransModeTransContr: 0 -> WOR Receiver (default)
TransModeEnableLBT : 0 -> Disabled (default)
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeEnabRepeat: 0 -> Disabled (default)
TransModeFixedTrans: 1 -> Fixed transmission (first three bytes can be used as high/low address and channel)
----------------------------------------
MODE NORMAL!
AUX HIGH!
Complete!
Save mode returned not recognized!
11
MODE CONFIGURATION!
AUX HIGH!
Complete!
3
Available buffer: 12 structure size: 12
AUX HIGH!
Complete!
----------------------------------------
HEAD : C1 0 9
 
AddH : 0
AddL : 3
NetID : 0
 
Chan : 23 -> 433MHz
 
SpeedParityBit     : 0 -> 8N1 (Default)
SpeedUARTDatte     : 11 -> 9600bps (default)
SpeedAirDataRate   : 10 -> 2.4kbps (default)
 
OptionSubPacketSett: 0 -> 240bytes (default)
OptionTranPower    : 0 -> 22dBm (Default)
OptionRSSIAmbientNo: 1 -> Enabled
 
TransModeWORPeriod : 11 -> 2000ms (default)
TransModeTransContr: 0 -> WOR Receiver (default)
TransModeEnableLBT : 0 -> Disabled (default)
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeEnabRepeat: 0 -> Disabled (default)
TransModeFixedTrans: 1 -> Fixed transmission (first three bytes can be used as high/low address and channel)
----------------------------------------
MODE NORMAL!
AUX HIGH!
Complete!
Success
1
----------------------------------------
HEAD : C1 0 9
 
AddH : 0
AddL : 3
NetID : 0
 
Chan : 23 -> 433MHz
 
SpeedParityBit     : 0 -> 8N1 (Default)
SpeedUARTDatte     : 11 -> 9600bps (default)
SpeedAirDataRate   : 10 -> 2.4kbps (default)
 
OptionSubPacketSett: 0 -> 240bytes (default)
OptionTranPower    : 0 -> 22dBm (Default)
OptionRSSIAmbientNo: 1 -> Enabled
 
TransModeWORPeriod : 11 -> 2000ms (default)
TransModeTransContr: 0 -> WOR Receiver (default)
TransModeEnableLBT : 0 -> Disabled (default)
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeEnabRepeat: 0 -> Disabled (default)
TransModeFixedTrans: 1 -> Fixed transmission (first three bytes can be used as high/low address and channel)
----------------------------------------
`

WillingLearner, 2021-06-17:
What might be the problem with it ? Any idea with debug output ?

xreef, 2021-07-06:
Hi Will,
some time return success sometime no, the library do the same work every time, so I think there is a connection problem, try to change pull up resistor.
Bye Renzo

closed 2021-07-06 by xreef

ricolk, 2025-09-17:
Hi, 
I ran into a similar issue and wanted to share a detail that might help others: I noticed that the IOs M0, M1, and AUX must be properly configured before powering up the E220 module. In my setup, the supply is controlled via a GPIO, which means IOs has been set after the module is already powered. This caused unexpected behavior during configuration.

`MODE NORMAL!
AUX HIGH!
Complete!
Save mode returned not recognized!
11
Available: 0`

Additionally, at startup, the AUX pin remained LOW and never rose after the module’s self-check, as described in Chapter 5.5 of the datasheet. 

> After the module is powered on, AUX will immediately output a low level, perform hardware self-test, and set the working mode according to user parameters; During this process, AUX remains low level. After completion, AUX outputs high level and starts working normally according to the working mode combined by M1 and M0;therefore, the user needs to wait for the rising edge of AUX as the starting point for the module to work normally.

Once I ensured the IOs were set before powering the module, everything worked as expected. 

`    pinMode(LORA_M0_PIN, OUTPUT);
    pinMode(LORA_M1_PIN, OUTPUT);
    pinMode(LORA_AUX_PIN, INPUT);
    digitalWrite(LORA_M0_PIN, LOW);
    digitalWrite(LORA_M1_PIN, LOW);`