xreef/EByte_LoRa_E22_Series_Library #5: Need  your help with transporting audio over Lora

happytm, 2021-08-11:
I have been using your great library for testing range of Ebyte E22 modules and it works perfectly.

I have following sketch which send sine wave data over ESPNow wireless link and playing it on other side via headphone connected to ESP32's internal DAC. It works fine.

Now I want to send same sine wave data using E22 modules (broadcast mode) instead of ESPNow for longer range. 

Is it possible ? If so can please show me example ?

My sketch is below:

```
#include <WiFi.h>
#include <esp_now.h>

#define lineIn ADC1_CHANNEL_0          // ADC 1 channel 0 GPIO36 for analog microphone connecttion
#define lineOut 25                     // DAC pin for headphone or amplified speaker connection
#define touchThreshold 50              // TTT (Touch To Transmit) threshold
#define audioBytes  240                // ESPNow allow upto 250 bytes packets

uint8_t audioData[audioBytes] = 
   
   // Sin wave
    {
    0x80, 0x83, 0x87, 0x8A, 0x8E, 0x91, 0x95, 0x98, 0x9B, 0x9E, 0xA2, 0xA5, 0xA7, 0xAA, 0xAD, 0xAF,
    0xB2, 0xB4, 0xB6, 0xB8, 0xB9, 0xBB, 0xBC, 0xBD, 0xBE, 0xBF, 0xBF, 0xBF, 0xC0, 0xBF, 0xBF, 0xBF,
    0xBE, 0xBD, 0xBC, 0xBB, 0xB9, 0xB8, 0xB6, 0xB4, 0xB2, 0xAF, 0xAD, 0xAA, 0xA7, 0xA5, 0xA2, 0x9E,
    0x9B, 0x98, 0x95, 0x91, 0x8E, 0x8A, 0x87, 0x83, 0x80, 0x7C, 0x78, 0x75, 0x71, 0x6E, 0x6A, 0x67,
    0x64, 0x61, 0x5D, 0x5A, 0x58, 0x55, 0x52, 0x50, 0x4D, 0x4B, 0x49, 0x47, 0x46, 0x44, 0x43, 0x42,
    0x41, 0x40, 0x40, 0x40, 0x40, 0x40, 0x40, 0x40, 0x41, 0x42, 0x43, 0x44, 0x46, 0x47, 0x49, 0x4B,
    0x4D, 0x50, 0x52, 0x55, 0x58, 0x5A, 0x5D, 0x61, 0x64, 0x67, 0x6A, 0x6E, 0x71, 0x75, 0x78, 0x7C
   };
  
// REPLACE WITH YOUR RECEIVER MAC Address
uint8_t broadcastAddress[] = {0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};

void OnDataRecv(const uint8_t * mac, const uint8_t *data, int len) {
  uint8_t audioIn;
  for (int i=0;i<len;i++) {
    audioIn += (uint8_t)data[i];
   dacWrite(lineOut, audioIn);
  }
}

void setup() {
  Serial.begin(115200);
  while(!Serial){};
  delay(500);
  Serial.println();

  // Set device as a Wi-Fi Station
  WiFi.mode(WIFI_STA);

  // Init ESP-NOW
  if (esp_now_init() != ESP_OK) {
    Serial.println("Error initializing ESP-NOW");
    return;
  }

  esp_now_register_recv_cb(OnDataRecv);
 
  // Register peer
  esp_now_peer_info_t peerInfo;
  memcpy(peerInfo.peer_addr, broadcastAddress, 6);
  peerInfo.channel = 0;  
  peerInfo.encrypt = false;
  
  // Add peer        
  if (esp_now_add_peer(&peerInfo) != ESP_OK){
    Serial.println("Failed to add peer");
    return;
  }

  Serial.println("Setup complete, move over to the Loop");
}
 
void loop() {
  
  if (touchRead(12) < touchThreshold )
  {     
       // Send audio packet via ESP-NOW
       esp_err_t result = esp_now_send(broadcastAddress, (const uint8_t *) audioData, audioBytes/*sizeof(audioData)*/); 
   }
} //End of Loop
```
Thanks.

xreef, 2021-08-12:
Hi happytm,
I think It's possible, you can find more information about broadcast message here.
https://www.mischianti.org/2021/01/28/ebyte-lora-e22-device-for-arduino-esp32-or-esp8266-library-part-2/#Fixed_transmission

To send
`ResponseStatus rs = e22ttl.sendBroadcastFixedMessage(0x17, "Message to a devices of a channel");`

To receive you must set broadcast address as own address
`configuration.ADDL = BROADCAST_ADDRESS;
configuration.ADDH = BROADCAST_ADDRESS;
`

Bye Renzo

closed 2021-08-12 by xreef

happytm, 2021-08-12:
Hi Renzo, 

I tried using ResponseStatus rs = e22ttl.sendBroadcastFixedMessage(15, audioData); but it takes String as an input for data. I need to send uint8_t audioData[240]  integer type array  which gives me an error.

Thanks.

xreef, 2021-08-16:
Send with structured message, you can check the tutorial of e32 that is the same thing.
Bye Renzo

happytm, 2021-08-16:
Thanks. I will try sending with structure.