xreef/EByte_LoRa_E22_Series_Library #8: This code only works 72 times with arduino

fixajteknik, 2022-01-16:
`#include "LoRa_E22.h"
#include <SoftwareSerial.h>

SoftwareSerial mySerial(10, 11);

/*
   Pinler     Arduino Nano    Lora E32 433T20d
                  11                3
                  10                4
*/

LoRa_E22 E22(&mySerial);

struct Signal {
  char type[15] = "Urfa63";
  byte role[4];
} data;

#define M0 7
#define M1 6
#define BTN 2
#define LED 9

byte i = 0;

void setup() {
  pinMode(M0, OUTPUT);
  pinMode(M1, OUTPUT);
  digitalWrite(M0, LOW);
  digitalWrite(M1, LOW);

  Serial.begin(9600);
  E22.begin();

}

void loop() {

  struct Signal {
    char type[4] = "Urfa";
    byte role[4];
  } data;

  i = i + 1;

  *(int*)( data.role) = i;

  ResponseStatus rs = E22.sendFixedMessage(0, 1, 25, &data, sizeof(Signal));
//  Serial.println(rs.getResponseDescription());
  Serial.print ("data: "); Serial.println(i);

  delay(1000);
}`

this works until i = 72 after that struct is loosing itselfs

Output:

`20:58:09.000 -> data: 1
20:58:10.118 -> data: 2
20:58:11.282 -> data: 3
20:58:12.404 -> data: 4
20:58:13.529 -> data: 5
20:58:14.649 -> data: 6
20:58:15.813 -> data: 7
20:58:16.933 -> data: 8
20:58:18.051 -> data: 9
20:58:19.215 -> data: 10
20:58:20.331 -> data: 11
20:58:21.453 -> data: 12
20:58:22.571 -> data: 13
20:58:23.740 -> data: 14
20:58:24.860 -> data: 15
20:58:25.976 -> data: 16
20:58:27.140 -> data: 17
20:58:28.258 -> data: 18
20:58:29.375 -> data: 19
20:58:30.510 -> data: 20
20:58:31.673 -> data: 21
20:58:32.791 -> data: 22
20:58:33.913 -> data: 23
20:58:35.031 -> data: 24
20:58:36.190 -> data: 25
20:58:37.308 -> data: 26
20:58:38.427 -> data: 27
20:58:39.589 -> data: 28
20:58:40.705 -> data: 29
20:58:41.824 -> data: 30
20:58:42.941 -> data: 31
20:58:44.106 -> data: 32
20:58:45.226 -> data: 33
20:58:46.344 -> data: 34
20:58:47.506 -> data: 35
20:58:48.623 -> data: 36
20:58:49.741 -> data: 37
20:58:50.906 -> data: 38
20:58:52.020 -> data: 39
20:58:53.140 -> data: 40
20:58:54.306 -> data: 41
20:58:55.424 -> data: 42
20:58:56.540 -> data: 43
20:58:57.704 -> data: 44
20:58:58.821 -> data: 45
20:58:59.935 -> data: 46
20:59:01.099 -> data: 47
20:59:02.217 -> data: 48
20:59:03.370 -> data: 49
20:59:04.492 -> data: 50
20:59:05.608 -> data: 51
20:59:06.727 -> data: 52
20:59:07.890 -> data: 53
20:59:09.009 -> data: 54
20:59:10.126 -> data: 55
20:59:11.291 -> data: 56
20:59:12.408 -> data: 57
20:59:13.529 -> data: 58
20:59:14.692 -> data: 59
20:59:15.811 -> data: 60
20:59:16.925 -> data: 61
20:59:18.090 -> data: 62
20:59:19.210 -> data: 63
20:59:20.328 -> data: 64
20:59:21.447 -> data: 65
20:59:22.609 -> data: 66
20:59:23.729 -> data: 67
20:59:24.846 -> data: 68
20:59:26.012 -> data: 69
20:59:27.129 -> data: 70
20:59:28.243 -> data: 71
20:59:29.406 -> data: 72
20:59:30.523 -> data: yUrfaI
20:59:31.639 -> da6
20:59:31.639 -> tzUrfaJ
`

fixajteknik, 2022-01-18:
ı think library need modifications. I have added free commend and fixed

`	memcpy(fixedStransmission->message,(unsigned char*)message,size);

	ResponseStatus status;
	status.code = this->sendStruct((uint8_t *)fixedStransmission, size+3);
	free(fixedStransmission);// ekleme
	if (status.code!=SUCCESS) return status;

//	free(fixedStransmission);`

xreef, 2022-01-20:
Thanks,
I start check all the examples for the e220 version.
Bye Renzo

fixajteknik, 2022-01-20:
Your welcome,İ have told at this video how to use your library in YouTubehttps://youtu.be/LqlZJ986tJIThanks to muchMehmet.

-- Yandex.Mail mobil uygulamasından gönderildi

xreef, 2022-01-20:
Hi,
It's beautiful. I'll try to share it.
Thanks Renzo

closed 2022-02-02 by xreef