xreef/EByte_LoRa_E22_Series_Library #11: configuration.TRANSMISSION_MODE.WORTransceiverControl = WOR_RECEIVER;

fixajteknik, 2022-10-24:
`configuration.TRANSMISSION_MODE.WORTransceiverControl = WOR_RECEIVER;`

this commend not effect lora module parameters setting for E22 onlt transmitter works

xreef, 2022-10-25:
Hi @fixajteknik,
It's very strange, the datasheet says that 1 is for the transmitter and 0 for the receiver.
I will try to test It as soon as possible.
Bye Renzo

fixajteknik, 2022-10-25:
also there is no `c.close();` at any set configration exampls. did you forget or `c.close` unneccesary

xreef, 2022-10-25:
Azz... no close is mandatory.
Thanks

xreef, 2022-12-01:
Hi @fixajteknik,
I try to redo some tests and seems It's working correctly without inverting the configuration.
Can you retry, please?
Bye Renzo

fixajteknik, 2023-01-09:
yes double check still reverse working.

also there is a new bug,

`  configuration.NETID = Netid;` is not save bigger number, it needs highbyte and lowbyte. also tested.

xreef, 2023-01-09:
Hi fixajteknik,
ok, I'm going to retest the TRANSMISSION_MODE.

NetId can be only one byte in the register (like ADDH and ADDL) It can't be more than that.

Thanks, Renzo

fixajteknik, 2023-01-16:
I have attached screenshoot, may be ı am making something wrong. on the other hand i didnt face any problem while working error translete or receiver.

[image]

[image]

xreef, 2023-01-17:
Hi,
mmm... I think that I remember that when I do the test I find that the bug is in the EByte software.

But your test says that It's the correct mine value of EByte software value.

Bye Renzo

xreef, 2023-02-21:
Hi @fixajteknik,
finally I receive the updated version from EByte

https://www.mischianti.org/forums/topic/fixed-ebyte-e22-desktop-program-for-usb-adapter/

Thanks Renzo

closed 2023-02-21 by xreef