xreef/EByte_LoRa_E22_Series_Library #13: Guru meditation error triggered within receiveStruct routine(s)

strud, 2022-11-09:
Hello,

I am using your library in a project that has an ESP32 in the master device and a Raspberry Pi PICO in the slave device.

The modules I am using are the E22-900T30S.

I am using the platformIO environment and have enabled the exception decoder to help diagnose the dump provided by the ESP32 when the Guru Meditation error occurs.

The the system operates for some time (say 5 minutes or more) where the master is sending packets to the slave and the slave responds. This is occuring at about 3  times per second.

When the crash occurs, this is the exception decode captured:

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400db8af  PS      : 0x00060b30  A0      : 0x80112d00  A1      : 0x3ffca3f0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000074  A5      : 0x3ffc1fc0
A6      : 0x00060120  A7      : 0x00000001  A8      : 0x800db8ac  A9      : 0x3ffca3d0  
A10     : 0x00000000  A11     : 0x00001800  A12     : 0x3f40bb9e  A13     : 0x00000003
A14     : 0x00000008  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4008626c  LEND    : 0x40086276  LCOUNT  : 0x00000000

Backtrace:0x400db8ac:0x3ffca3f00x40112cfd:0x3ffca410 0x400da07d:0x3ffca430 0x400da4f0:0x3ffca450 0x400da59c:0x3ffca490 0
x400d6533:0x3ffca4c0 

  #0  0x400db8ac:0x3ffca3f0 in Stream::readBytes(char*, unsigned int) at C:/Users/T520/.platformio/packages/framework-ar
duinoespressif32/cores/esp32/Stream.cpp:286
  #1  0x40112cfd:0x3ffca410 in Stream::readBytes(unsigned char*, unsigned int) at C:/Users/T520/.platformio/packages/fra
mework-arduinoespressif32/cores/esp32/Stream.h:103
  #2  0x400da07d:0x3ffca430 in LoRa_E22::receiveStruct(void*, unsigned short) at lib/EByte_LoRa_E22_Series_Library/LoRa_
E22.cpp:463
  #3  0x400da4f0:0x3ffca450 in LoRa_E22::receiveMessageComplete(unsigned char, bool) at lib/EByte_LoRa_E22_Series_Librar
y/LoRa_E22.cpp:781
  #4  0x400da59c:0x3ffca490 in LoRa_E22::receiveMessageRSSI(unsigned char) at lib/EByte_LoRa_E22_Series_Library/LoRa_E22
.cpp:774
  #5  0x400d6533:0x3ffca4c0 in Task_RadioLink(void*) at src/main.cpp:558

As you can see, the top level call is to receiveMessageRSSI()

I have a struct that I am passing and this is 116 bytes long. 

I was using the following in my call;

`rsc = e22ttl.receiveMessageRSSI(sizeof(padControllerData));`

but due to this problem occuring I thought I should try a dedicated variable set to this size 

`rsc = e22ttl.receiveMessageRSSI(padControllerDataSize);`   where padControllerDataSize is defined as an unsigned char

This has had no effect.

I then looked into your library and found the call I believe is causing this fault:  (I added some debug prints to help isolate the issue)

`Status LoRa_E22::receiveStruct(void *structureManaged, uint16_t size_) {
	Status result = E22_SUCCESS;

	DEBUG_PRINT(" structure size: ");		// added CS 9/11
	DEBUG_PRINTLN(size_);					// added CS 9/11
	DEBUG_PRINT("calling readBytes ");      // added CS 9/11

	uint8_t len = this->serialDef.stream->readBytes((uint8_t *) structureManaged, size_);

	DEBUG_PRINT("Available buffer: ");
	DEBUG_PRINT(len);
/*	DEBUG_PRINT(" structure size: ");
	DEBUG_PRINTLN(size_);
*/
	if (len!=size_){
		if (len==0){
			result = ERR_E22_NO_RESPONSE_FROM_DEVICE;
		}else{
			result = ERR_E22_DATA_SIZE_NOT_MATCH;
		}
	}
	if (result != E22_SUCCESS) return result;

	result = this->waitCompleteResponse(1000);
	if (result != E22_SUCCESS) return result;

	return result;
}
`

The last debug just prior to the Guru Meditation error is:

 structure size: 116
calling readBytes Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

So this shows the error is occuring on the call;

`uint8_t len = this->serialDef.stream->readBytes((uint8_t *) structureManaged, size_);`

Is there something that can be done to ensure the pointer 'structureManaged' is valid before making this call to readBytes?

Super keen to get assistance on this as my project is not stable or usable with this error.

strud, 2022-11-09:
So looking at this again, I notice malloc is being used to allocate memory for the pointer to hold the structure being passed.

I could not find anywhere in the code that this allocated memory was being freed, so I added this to my code after reading the structure.

I then looked at the definition of rsc in the library and noted there is an inline function called close() that does this too, however this was not being called in your example for receiving a structure.

I am now running a test with the allocated memory being freed to see if this solves the problem (I am hopeful).

xreef, 2022-11-12:
Hi Strud,
sorry if I answer only now, yes, you must use the .close() function embedded in the ResponseStructure.
Bye Renzo

closed 2022-11-12 by xreef