xreef/EByte_LoRa_E22_Series_Library #16: FREQUENCY_915 is incorrect

berk-ott, 2022-12-27:
Thank you in advance!

'OPERATING_FREQUENCY' is set to 850 
I set it to 915 in my copy of statesNaming.h
But I am not sure if that is correct.
Is it supposed to be center of the band?

#elif defined(FREQUENCY_868)
	#define OPERATING_FREQUENCY 850
#elif defined(FREQUENCY_915)
>>>>>	// This is incorrect?  #define OPERATING_FREQUENCY 850. <<<<<
	#define OPERATING_FREQUENCY 915 // LORA_HBO 2022-12-27
#else
	#define OPERATING_FREQUENCY 410
#endif

https://www.baranidesign.com/faq-articles/2019/4/23/lorawan-usa-frequencies-channels-and-sub-bands-for-iot-devices

berk-ott, 2022-12-27:
#elif defined(FREQUENCY_915)
	// This is incorrect?  #define OPERATING_FREQUENCY 850
	#define OPERATING_FREQUENCY 902 // LORA_HBO 2022-12-27

I actually set it to 902

setConfiguration reports
Chan : 0 -> 902MHz

xreef, 2022-12-29:
Hi Berk,
the LoRa 915Mhz device no more exists, now is substituted with 868Mhz that cove also 900Mhz frequencies.
bye Renzo

closed 2022-12-29 by xreef