xreef/EByte_LoRa_E22_Series_Library #19: No support for 0.3k and 1k2 modes in E22 modules V2.1 (and V2.0)

OK2MOP, 2023-01-11:
Hello, this is not a bug in library itself, just a warning and maybe suggestion for enhancement (adding the check or documentation). The new V2.1 modules (PCB versions 10498-V1.1 and 10497-V1.0)  - and after confirming from EBYTE also the V2.0 modules - do not correctly set up speeds 0.3k and 1k2 and silently ignore the setting (they probably stay at the 2k4 speed like E220s). It seems the SX127x has been replaced in them by LLCC68 inside, which does not support spreading factors needed by these modes. 

They can be identified by querying register 0x82 (C1 82 01), which returns 0x40 instead of 0x10 or 0x11 seen in previous versions. Older modules set to these speeds cannot communicate with the V2.1 ones on these speeds and vice versa. 

I had this just confirmed from EBYTE, my guess is that they are in fact re-labeled E220-400T30D/E220-900T30D modules.

So my suggestion is adding a warning to README.md concerning those speeds.

xreef, 2023-01-13:
Hi OK2MOP,
thanks a lot for the tips, if you want you can do a pull request with the fix of the README.
Thanks Renzo

fixajteknik, 2023-01-16:
Thanks for detailed information OK2MOP also same issue happening E32 lora modüles. Old version and new versions cannot communicate each other until up to air rate 9.6kbps.

onurcansari, 2024-06-30:
[image]

Merhaba,
Yukarıda fotoğrafta belirttiğim iki adet LoRa modülüm var: biri normal E22, diğeri ise E22 V2. İki adet RP2040 işlemciye sahip bir kart tasarladım. Bu kartlardan normal E22'ye sahip olanını seri port üzerinden programlayabiliyorum. Ancak, E22 V2'ye sahip olanını programlamaya çalıştığımda sadece "FFFFFFF" geri döndürüyor. İkisini de aynı kod ile programlamaya çalışıyorum. Bu konuda yardımcı olabilir misiniz?

onurcansari, 2024-06-30:
[image]

Hello,
In the photo above, I have two LoRa modules: one is a standard E22, and the other is an E22 V2. I designed a board with two RP2040 processors. I can program the board with the standard E22 through serial print, but when I try to program the one with the E22 V2, it only returns "FFFFFFF". I am using the same code to program both. Can you help me with this issue?

closed 2024-07-16 by xreef

tritile, 2025-05-07:
I've noticed the same issue in E32 433T30D modules with board versions 10408-V1.1 and 10498-V1.1.

10498 boards will allow air rate register to be setted to values bellow 2400, but will transmit in 2400 anyway.
10408 boards won't even allow register to be setted to anything bellow 2400.

I have documented it here: https://github.com/KrisKasprzak/EBYTE/issues/84