xreef/EByte_LoRa_E22_Series_Library #22: Não estou conseguindo colocar a frequência de comunicação em 915Mhz, so trabalha no padrão 433Mhz

VitalFilho, 2023-03-06:
não estou conseguindo mudar o Canal de trabalho para a frequência 915Mhz, sendo assim não tem referencia para a frequência 915Mhz e não sei qual o canal utilizar sem ser o CHAN 23 que e referente ao canal 433Mhz

xreef, 2023-03-06:
Ciao,
trovi la guida agli E22 [qui](https://www.mischianti.org/it/category/le-mie-librerie/dispositivi-ebyte-lora-e22/).
Ciao RM

closed 2023-03-06 by xreef

VitalFilho, 2023-03-06:
Não estou conseguindo encontrar os canais adequados na sua biblioteca para utilização da frequência 915Mhz todos os arquivos são para 433Mhz

xreef, 2023-03-06:
https://www.mischianti.org/it/2021/01/28/ebyte-lora-e22-per-arduino-esp32-o-esp8266-libreria-2/#Potenza_trasmissione

https://www.mischianti.org/forums/topic/esp8266-wemos-d1-mini-and-e220-900t30d-frequencies-problem-and-info/