xreef/EByte_LoRa_E22_Series_Library #25: NodeMCU wiring

Kamoba, 2023-12-15:
Hi,

Thank you for your great work.

I'm using E22-900T30S with NodeMCU V3, so I guess I should use the wiring and code for esp8266.
Here is my wiring:
```
E22	esp8266
------------------
M0	D7
M1	D6
TX	PIN D2 (PullUP 4,7KΩ)
RX	PIN D3 (PullUP 4,7KΩ)
AUX     PIN D5 (PullUP 4,7KΩ)
VCC	3.3v
GND	GND
```
To get started I use the [esp8266_e22_getConfiguration](https://github.com/xreef/EByte_LoRa_E22_Series_Library/blob/master/examples/esp8266_e22_getConfiguration/esp8266_e22_getConfiguration.ino) code.
But got error:
```
No response from device! (Check wiring)
12
----------------------------------------
HEAD : 0 0 0
 
AddH : 0
AddL : 0
NetID : 0
 
Chan : 0 -> 410MHz
 
SpeedParityBit     : 0 -> 8N1 (Default)
SpeedUARTDatte     : 0 -> 1200bps
SpeedAirDataRate   : 0 -> 0.3kbps
 
OptionSubPacketSett: 0 -> 240bytes (default)
OptionTranPower    : 0 -> 22dBm (Default)
OptionRSSIAmbientNo: 0 -> Disabled (default)
 
TransModeWORPeriod : 0 -> 500ms
TransModeTransContr: 0 -> WOR Receiver (default)
TransModeEnableLBT : 0 -> Disabled (default)
TransModeEnableRSSI: 0 -> Disabled (default)
TransModeEnabRepeat: 0 -> Disabled (default)
TransModeFixedTrans: 0 -> Transparent transmission (default)
----------------------------------------
No response from device! (Check wiring)
12
----------------------------------------
HEAD: 0 0 0
Model no.: 61
Version  : 72
Features : 65
----------------------------------------
```

Did I do something wrong? Or NodeMCU if supported use different wiring?

Thanks

Kamoba, 2023-12-16:
Ok, I got success. The trick was : 
```
M0  -->  GND
M1  -->  NC/VCC
RX  -->  TX2 (D7)
TX  -->  RX2 (D8)
```

closed 2023-12-16 by Kamoba

xreef, 2023-12-16:
Hi,
for the frequency you must set the specified define.
You cancheck [here](https://mischianti.org/ebyte-lora-e220-llcc68-device-for-arduino-esp32-or-esp8266-library-2/).
Bye Renzo

Kamoba, 2023-12-18:
Hmm, I add this:
```
#define E22_30
#define FREQUENCY_915
```
But still got ```Chan : 63 -> 473MHz``` while I have Ch: 63 and and Frequency: 913.125 MHz.
On the setConfig script I change pins and ```configuration.CHAN = 63;``` but got same thing.

xreef, 2023-12-18:
Hi,
First of all remember that only the description is wrong.
To put on work if the define not working insert them inside stateNaming.h file.
Bye Renzo

Kamoba, 2023-12-18:
Hi, 
I was scaring to transmit on a wrong frequency bands and got trouble. So thank you for taking the stress away.

After a few tries, I got it to work! So to summarize, I skip your configuration and use the configuration software provided by the manufacturer ([RF_settings](https://www.cdebyte.com/products/E22-900T30S/4#Downloads)) then use your fixedTransmission example.

A suggestion, you should set variable CHANNEL on the example
 ```e22ttl.sendFixedMessage(0, DESTINATION_ADDL, CHANNEL, "Hello, world?");``` I almost did not pay attention to it.

A last question could you please explain me what is corresponding AddL here on this image?

[image]

You said: "Low address byte of module (the default 00H)", but still did not get, yes I know I'm a newbie.

Thanks!

xreef, 2023-12-20:
Hi koomba,
ADDL and ADDH are the bytes that represent the adddress I use it separately for simplicity.
Bye Renzo