xreef/EByte_LoRa_E22_Series_Library #27: How to add sender and receiver addresses? Using ESP32

JayZeeM, 2024-01-16:


JayZeeM, 2024-01-16:
This is for a communication device that connects each other without LoRaWAN or internet.

xreef, 2024-01-16:
Hi jay,
You can add the sender address in the payload, the receiver it's already present in the device.
Bye Renzo

JayZeeM, 2024-01-16:
I mean like where will I input the receiver's address in the sending end? Can you provide and example?

JayZeeM, 2024-01-17:
Do you also have a project of connecting an ESP32 to a phone passing data through an application

xreef, 2024-01-18:
Hi Jay,
To add the address, you need to manage the structure you send and put it inside a managed position.
To communicate your ESP32 with cellular, you can use Bluetooth, which can work like a wireless UART.
But I don't have any examples.
Bye Renzo

closed 2024-02-06 by JayZeeM