xreef/EByte_LoRa_E22_Series_Library #28: Help

JayZeeM, 2024-02-06:
[image]

Hello im trying to use this code for the receiver with the connections above

#define E22_30
#define FREQUENCY_868

#include "Arduino.h"
#include "LoRa_E22.h"
 
LoRa_E22 e22ttl(3, 1, &Serial2, 18, 21, 19, UART_BPS_RATE_9600); //  esp32 RX <-- e22 TX, esp32 TX --> e22 RX AUX M0 M1
 
void setup() {
  Serial.begin(9600);
  delay(500);
 
  Serial.println("Hi, I'm going to send message!");
 
  // Startup all pins and UART
  e22ttl.begin();

  	ResponseStructContainer c;
	c = e22ttl.getConfiguration();
	// It's important get configuration pointer before all other operation
	Configuration configuration = *(Configuration*) c.data;
	Serial.println(c.status.getResponseDescription());
	Serial.println(c.status.code);

  configuration.ADDL = 0x18; // First part of address
  configuration.ADDH = 0x00; // Second part
  configuration.NETID = 0x00; // NETID used for repeater function
  configuration.CHAN = 18;
  configuration.OPTION.transmissionPower = POWER_24;
}
 
void loop() {
    // If something available
  if (e22ttl.available()>1) {
      // read the String message
    ResponseContainer rc = e22ttl.receiveMessage();
    // Is something goes wrong print error
    if (rc.status.code!=1){
        rc.status.getResponseDescription();
    }else{
        // Print the data received
        Serial.println(rc.data);
    }
  }
  if (Serial.available()) {
      String input = Serial.readString();
      e22ttl.sendMessage(input);
  }
  delay(5000);
}

JayZeeM, 2024-02-06:
and this for the sender 

#define E22_30
#define FREQUENCY_868

#include "Arduino.h"
#include "LoRa_E22.h"
 
LoRa_E22 e22ttl(3, 1, &Serial2, 18, 21, 19, UART_BPS_RATE_9600); //  esp32 RX <-- e22 TX, esp32 TX --> e22 RX AUX M0 M1

void setup() {
  Serial.begin(9600);
  delay(500);
 
  Serial.println("Hi, I'm going to send message!");
 
  // Startup all pins and UART
  e22ttl.begin();
 
	ResponseStructContainer c;
	c = e22ttl.getConfiguration();
	// It's important get configuration pointer before all other operation
	Configuration configuration = *(Configuration*) c.data;
	Serial.println(c.status.getResponseDescription());
	Serial.println(c.status.code);

  configuration.ADDL = 0x18; // First part of address
  configuration.ADDH = 0x00; // Second part
  configuration.NETID = 0x00; // NETID used for repeater function
  configuration.CHAN = 18;
  configuration.OPTION.transmissionPower = POWER_24;

}
 
void loop() {

  // Send message
  ResponseStatus rs = e22ttl.sendMessage("Hello, world?");
  // Check If there is some problem of successfully send
  Serial.println(rs.getResponseDescription());

  delay(10000);

}

but the receiver isn't showing anything only this 
[image]

xreef, 2024-02-06:
Hi Jay,
check that all the values of the configuration are correct.
Bye Renzo

closed 2024-07-19 by xreef