xreef/EByte_LoRa_E22_Series_Library #29: Impossible to use the library

18o61914, 2024-07-18:
Good evening, dear Renzo! I am writing this issue after 20 hours of fighting with your library to tell you that it is very hard to manage it. I've learned the "documentation" from the first letter to the last one, I've read all your methods and structures (indeed the ideology of libraries is oposite to this actions, but you forced me to look at your code), but I still can not to configurate my E22, couse of "Save mode returned not recognized! 11" issue. Os course I whant to say a huge thanks to you for you have made this project, it's a really good thing. And I also whant to sugest you to stop writing spagetti code or to use defines, that was defined in files of 105 nesting level, define them .h file or at least name them in documentation, couse it is not normal to get ==11== error code and to see:

```
if (WRONG_FORMAT == ((Configuration *)&configuration)->COMMAND){
rc.code = ERR_E22_WRONG_FORMAT;
}
if (RETURNED_COMMAND != ((Configuration *)&configuration)->COMMAND || REG_ADDRESS_CFG!= ((Configuration *)&configuration)->STARTING_ADDRESS || PL_CONFIGURATION!= ((Configuration *)&configuration)->LENGHT){
rc.code = ERR_E22_HEAD_NOT_RECOGNIZED;
}
```

in the code! Also whant to sugest you to make a normal documentation in your future projects and to think about users of your lib, while  making the project.

Best wishes from the user of your programming product, if you found the answer on "Save mode returned not recognized! 11" - give the responce, please!

xreef, 2024-07-19:
Hi 18o61914,
Your opinion is interesting, and I hope that you can write better code and a library. I look forward to your library on GitHub with better documentation.
In the meantime, I advise you to read the forum topic and issue and find the problem that you explained because other people had the same issue.
Bye Renzo

closed 2024-07-19 by xreef

18o61914, 2024-07-19:
UPD, I've turned on my analyzer and found out, that your lib didn't make HIGH level on M0 and M1 pins, so I've just made my own sketch to configure the module. 

```
#include "Arduino.h"

#define     M0      23
#define     M1      22
#define     AUX     21

void setup() {
  Serial.begin(9600);
  Serial2.begin(9600);

  pinMode(M0, OUTPUT);
  pinMode(M1, OUTPUT);
  pinMode(AUX, INPUT);

  digitalWrite(M0, HIGH);
  digitalWrite(M1, HIGH);
  delay(5000);
  char buff[3] = {0xC1, 0x00, 0x08};
  char buff2[4] = {0xC0, 0x02, 0x01, 0b01100000};
  char buff3[4] = {0xC0, 0x03, 0x01, 0b11000000};
  Serial2.print(buff[0]);
  Serial2.print(buff[1]);
  Serial2.print(buff[2]);
  delay(1000);
  Serial2.print(buff2[0]);
  Serial2.print(buff2[1]);
  Serial2.print(buff2[2]);
  Serial2.print(buff2[3]);
  delay(1000);
  Serial2.print(buff3[0]);
  Serial2.print(buff3[1]);
  Serial2.print(buff3[2]);
  Serial2.print(buff3[3]);
  delay(1000);
  Serial2.print(buff[0]);
  Serial2.print(buff[1]);
  Serial2.print(buff[2]);
}

void loop() {
  // put your main code here, to run repeatedly:

}
```
There is my code, if someone need to configure the module. 
The link for the documentation: https://www.ebyte.com/en/downpdf.aspx?id=1212
Page 13
7.1-7.3 chapters

18o61914, 2024-07-19:
Renzo, I'm sorry if I hurt your feelings as the developer of this library. I didn't mean to devalue the work you've done or make it seem like I'm a better programmer than you. Of course, I won’t be able to write a library due to lack of experience, time and motivation, so don’t think that I wanted to offend you or anything. I even thanked you for your work in the first comment. Consider my negative review as a review from a person who used your software product, but encountered some obstacles that decided to tell you about in order to improve your software products. Once again, thank you for the library, I still haven’t seen a better library than this one. Best wishes!